fpoussin/QStlink2 #17: 'msleep' is a protected member of 'QThread' with Qt4 4.8.7_1@mac 10.11

monkeyxite, 2015-11-06:
As title

```
➜  QStlink2 git:(master) ✗ qmake
Project MESSAGE: Building version 1.2.1 for Qt 4.8.7
Project MESSAGE: Building QtUsb with LibUsb 1.0 support.
➜  QStlink2 git:(master) ✗ make
clang++ -c -pipe -stdlib=libc++ -mmacosx-version-min=10.7 -O2 -arch x86_64 -Wall -W -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/local/Cellar/qt/4.8.7_1/mkspecs/unsupported/macx-clang-libc++ -I. -I/usr/local/Cellar/qt/4.8.7_1/lib/QtCore.framework/Versions/4/Headers -I/usr/local/Cellar/qt/4.8.7_1/lib/QtCore.framework/Versions/4/Headers -I/usr/local/Cellar/qt/4.8.7_1/lib/QtGui.framework/Versions/4/Headers -I/usr/local/Cellar/qt/4.8.7_1/lib/QtGui.framework/Versions/4/Headers -I/usr/local/Cellar/qt/4.8.7_1/lib/QtXml.framework/Versions/4/Headers -I/usr/local/Cellar/qt/4.8.7_1/lib/QtXml.framework/Versions/4/Headers -I/usr/local/Cellar/qt/4.8.7_1/include -Iinc -IQtUsb/src -I. -I. -F/usr/local/Cellar/qt/4.8.7_1/lib -o main.o src/main.cpp
src/main.cpp:174:62: error: 'msleep' is a protected member of 'QThread'
                while (w->mTfThread->isRunning()) { QThread::msleep(100); }
                                                             ^
/usr/local/Cellar/qt/4.8.7_1/lib/QtCore.framework/Versions/4/Headers/qthread.h:116:17: note: declared protected here
    static void msleep(unsigned long);
                ^
src/main.cpp:178:62: error: 'msleep' is a protected member of 'QThread'
                while (w->mTfThread->isRunning()) { QThread::msleep(100); }
                                                             ^
/usr/local/Cellar/qt/4.8.7_1/lib/QtCore.framework/Versions/4/Headers/qthread.h:116:17: note: declared protected here
    static void msleep(unsigned long);
                ^
src/main.cpp:182:62: error: 'msleep' is a protected member of 'QThread'
                while (w->mTfThread->isRunning()) { QThread::msleep(100); }
                                                             ^
/usr/local/Cellar/qt/4.8.7_1/lib/QtCore.framework/Versions/4/Headers/qthread.h:116:17: note: declared protected here
    static void msleep(unsigned long);
                ^
src/main.cpp:184:22: error: 'msleep' is a protected member of 'QThread'
            QThread::msleep(300);
                     ^
/usr/local/Cellar/qt/4.8.7_1/lib/QtCore.framework/Versions/4/Headers/qthread.h:116:17: note: declared protected here
    static void msleep(unsigned long);
                ^
4 errors generated.
make: *** [main.o] Error 1
```

monkeyxite, 2015-11-06:
subclass QThread to expose msleep methods in main and mainwindow for qt4

```C++
class SleepThread : public QThread {
public: 
   static inline void msleep(unsigned long msecs) { 
       QThread::msleep(msecs);
   }
};
```

fpoussin, 2015-11-10:
Thanks for finding that, I'll commit your fix.

closed 2015-11-11 by fpoussin