fpoussin/QStlink2 #18: Support for STM32L series

wrlear, 2015-11-08:
Any chance you will add support for STM32L processors? Especially the L053 and L152 supported by Nucleo boards

Thanks,
Bill

fpoussin, 2015-11-10:
Can you provide some debug output?

wrlear, 2015-11-16:
Sorry for the delayed response. Fighting other fires. I'll check debug output the next time I can get back to this project. 
Bill

wrlear, 2015-12-01:
Sorry for my lack of response. We replaced the L0 with an L1, and I have not been able to get back to this to recreate the problem.

I’ll try to reconnect the Nucleo L0 tomorrow and get more info for you.

 

Bill

fpoussin, 2015-12-11:
Did you have time to retry?

wrlear, 2015-12-11:
Fabien,

Again, I apologize for my delayed responses – Attached are the debug outputs from two tests. qstlink2 failed to connect to the NUCLEO_STM32L053  (stm32l053r8t6)  , but succeeded with a NUCLEO_STM32L152 (stm32l152ret6).

Both boards work with OpenOCD and with the windows stlink program

 

The tests were run on Ubuntu14.04 running in a VirtualBox VM on a Windows8.1 host.  I installed qstlink2 from your ppa:

$ sudo add-apt-repository ppa:fpoussin/ppa

$ sudo apt-get update

$ sudo apt-get install qstlink2

 

 

It shows it is version 1.2.0

 

I hope this helps,

Bill

 

Nucleo STM32L152
bill@bill-ubuntu-vm:~$ qstlink2 -v
Verbose level: 5 
156 - Debug: Loading device list.
157 - Info: Devices list loaded. 
158 - Debug: "RDPTR_KEY" -> 165 
158 - Debug: "KEY1" -> 1164378403 
158 - Debug: "KEY2" -> 3455027627 
158 - Debug: "OPTKEY1" -> 135866939 
158 - Debug: "OPTKEY2" -> 1281191551 
158 - Debug: "SR_BSY" -> 0 
158 - Debug: "SR_PER" -> 2 
158 - Debug: "SR_EOP" -> 5 
158 - Debug: "CR_PG" -> 0 
158 - Debug: "CR_PER" -> 1 
158 - Debug: "CR_MER" -> 2 
158 - Debug: "CR_STRT" -> 6 
158 - Debug: "CR_LOCK" -> 7 
158 - Debug: "CR_PGSIZE" -> 8 
158 - Debug: "ACR_OFFSET" -> 0 
158 - Debug: "KEYR_OFFSET" -> 4 
158 - Debug: "OPT_KEYR_OFFSET" -> 8 
158 - Debug: "SR_OFFSET" -> 12 
158 - Debug: "CR_OFFSET" -> 16 
158 - Debug: "AR_OFFSET" -> 20 
158 - Debug: "OBR_OFFSET" -> 28 
158 - Debug: "WRPR_OFFSET" -> 32 
158 - Debug: "devices_default" -> "flash_base" 
158 - Debug: "devices_default" -> "sram_base" 
158 - Debug: "devices_default" -> "buffer_size" 
158 - Debug: "devices_default" -> "flash_size" 
158 - Debug: "devices_default" -> "flash_pgsize" 
158 - Debug: "device" -> "STM32L03xx" -> "core_id" 
158 - Debug: "device" -> "STM32L03xx" -> "chip_id" 
158 - Debug: "device" -> "STM32L03xx" -> "flash_size_reg" 
158 - Debug: "device" -> "STM32L03xx" -> "flash_int_reg" 
158 - Debug: "device" -> "STM32L03xx" -> "buffer_size" 
158 - Debug: "device" -> "STM32L03xx" -> "loader" 
158 - Debug: "device" -> "STM32L05xx" -> "core_id" 
158 - Debug: "device" -> "STM32L05xx" -> "chip_id" 
[... 527 lines cut ...]
4133 - Info: "Fetching version..." 
4133 - Debug: ***[ stlinkv2::STVersion stlinkv2::getVersion() ]*** 
4156 - Debug: ***[ void stlinkv2::setExitModeDFU() ]*** 
4164 - Info: "Changing mode to SWD..." 
4164 - Debug: ***[ void stlinkv2::setModeSWD() ]*** 
4164 - Debug: ***[ quint8 stlinkv2::getMode() ]*** 
4173 - Debug: ***[ void stlinkv2::setExitModeDFU() ]*** 
4282 - Info: "Fetching mode..." 
4283 - Debug: ***[ quint8 stlinkv2::getMode() ]*** 
4289 - Info: "Mode: Debug" 
4289 - Info: "Fetching status..." 
4289 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
4298 - Info: "Status: Core Running" 
4298 - Info: "Fetching MCU Info..." 
4298 - Debug: ***[ quint32 stlinkv2::getCoreID() ]*** 
4306 - Info: CoreID: "bc11477" 
4306 - Debug: ***[ void stlinkv2::resetMCU() ]*** 
4382 - Debug: ***[ quint32 stlinkv2::getChipID() ]*** 
4382 - Debug: ***[ qint32 stlinkv2::readMem32(quint32, quint16) ]***  Reading at "0xE0042000" 
4390 - Info: CM3/4 Searching at "e0042000" 
4390 - Info: ChipID: "0" 
4390 - Debug: Looking for: "0" 
4390 - Error: Did not find chipID! 
4390 - Info: "Device not found in database!" 
4391 - Error: Device not found in database!

fpoussin, 2015-12-11:
Hi,

You should have V1.2.1 from the PPA.
Can you check why the version is incorrect? Maybe you are using a proxy.

wrlear, 2015-12-11:
Ok, I just did an apt-get update and re-installed qstlink2.  I now have version 1.2.1  (I had done the previous install prior to first reporting the problem).

Unfortunately, I get the same problem (see attached output).  qstlink2 gets what appears to be a valid CoreID, but gets a ChipID of 0x00

 

Bill

 

177 - Debug: Loading device list.
178 - Info: Devices list loaded.
180 - Debug: "RDPTR_KEY" -> 165 
180 - Debug: "KEY1" -> 1164378403 
180 - Debug: "KEY2" -> 3455027627 
181 - Debug: "OPTKEY1" -> 135866939 
181 - Debug: "OPTKEY2" -> 1281191551 
181 - Debug: "SR_BSY" -> 0 
181 - Debug: "SR_PER" -> 2 
181 - Debug: "SR_EOP" -> 5 
181 - Debug: "CR_PG" -> 0 
181 - Debug: "CR_PER" -> 1 
181 - Debug: "CR_MER" -> 2 
181 - Debug: "CR_STRT" -> 6 
181 - Debug: "CR_LOCK" -> 7 
181 - Debug: "CR_PGSIZE" -> 8 
181 - Debug: "ACR_OFFSET" -> 0 
181 - Debug: "KEYR_OFFSET" -> 4 
181 - Debug: "OPT_KEYR_OFFSET" -> 8 
181 - Debug: "SR_OFFSET" -> 12 
181 - Debug: "CR_OFFSET" -> 16 
181 - Debug: "AR_OFFSET" -> 20 
181 - Debug: "OBR_OFFSET" -> 28 
181 - Debug: "WRPR_OFFSET" -> 32 
181 - Debug: "devices_default" -> "flash_base" 
181 - Debug: "devices_default" -> "sram_base" 
181 - Debug: "devices_default" -> "buffer_size" 
181 - Debug: "devices_default" -> "flash_size" 
181 - Debug: "devices_default" -> "flash_pgsize" 
181 - Debug: "device" -> "STM32L03xx" -> "core_id" 
181 - Debug: "device" -> "STM32L03xx" -> "chip_id" 
181 - Debug: "device" -> "STM32L03xx" -> "flash_size_reg" 
181 - Debug: "device" -> "STM32L03xx" -> "flash_int_reg" 
181 - Debug: "device" -> "STM32L03xx" -> "buffer_size" 
181 - Debug: "device" -> "STM32L03xx" -> "loader" 
181 - Debug: "device" -> "STM32L05xx" -> "core_id" 
181 - Debug: "device" -> "STM32L05xx" -> "chip_id" 
182 - Debug: "device" -> "STM32L05xx" -> "flash_size_reg" 
182 - Debug: "device" -> "STM32L05xx" -> "flash_int_reg" 
182 - Debug: "device" -> "STM32L05xx" -> "buffer_size" 
182 - Debug: "device" -> "STM32L05xx" -> "loader" 
182 - Debug: "device" -> "STM32L07xx" -> "core_id" 
182 - Debug: "device" -> "STM32L07xx" -> "chip_id" 
182 - Debug: "device" -> "STM32L07xx" -> "flash_size_reg" 
182 - Debug: "device" -> "STM32L07xx" -> "flash_int_reg" 
182 - Debug: "device" -> "STM32L07xx" -> "buffer_size" 
182 - Debug: "device" -> "STM32L07xx" -> "loader" 
182 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "core_id" 
182 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "chip_id" 
182 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "flash_size_reg" 
182 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "flash_int_reg" 
182 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "buffer_size" 
182 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "loader" 
182 - Debug: "device" -> "STM32L1xx (High Density)" -> "core_id" 
183 - Debug: "device" -> "STM32L1xx (High Density)" -> "chip_id" 
183 - Debug: "device" -> "STM32L1xx (High Density)" -> "flash_size_reg" 
183 - Debug: "device" -> "STM32L1xx (High Density)" -> "flash_int_reg" 
183 - Debug: "device" -> "STM32L1xx (High Density)" -> "buffer_size" 
183 - Debug: "device" -> "STM32L1xx (High Density)" -> "loader" 
183 - Debug: "device" -> "STM32L1xx cat2" -> "core_id" 
[... 212 lines cut ...]
4452 - Info: "Changing mode to SWD..." 
4452 - Debug: ***[ void stlinkv2::setModeSWD() ]*** 
4452 - Debug: ***[ quint8 stlinkv2::getMode() ]*** 
4457 - Debug: ***[ void stlinkv2::setExitModeDFU() ]*** 
4564 - Info: "Fetching mode..." 
4564 - Debug: ***[ quint8 stlinkv2::getMode() ]*** 
4567 - Info: "Mode: Debug" 
4567 - Info: "Fetching status..." 
4567 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
4571 - Info: "Status: Core Running" 
4571 - Info: "Fetching MCU Info..." 
4571 - Debug: ***[ quint32 stlinkv2::getCoreID() ]*** 
4575 - Info: CoreID: "bc11477" 
4575 - Debug: ***[ void stlinkv2::resetMCU() ]*** 
4650 - Debug: ***[ quint32 stlinkv2::getChipID() ]*** 
4650 - Debug: ***[ qint32 stlinkv2::readMem32(QByteArray*, quint32, quint16) ]*** "Reading at E0042000" 
4653 - Info: CM3/4 Searching at E0042000
4653 - Info: ChipID: 0x000
4653 - Debug: Looking for: 0x000
4653 - Error: Did not find chipID!
4654 - Info: "Device not found in database!" 
4654 - Error: Device not found in database!
4654 - Info: "Disconnecting..." 
4657 - Debug: Closing USB connection...
4657 - Info: "Disconnected."

fpoussin, 2015-12-11:
I fixed the problem (in theory), please update with apt in a few minutes.

wrlear, 2015-12-11:
Ahh, much better!  V1.2.2 now correctly reports ChipID 0x417, device type STM32L05xx with 64KB flash.

 

Thanks.  

Bill

 

Version: 1.2.2
195 - Debug: Loading device list.
199 - Info: Devices list loaded.
200 - Debug: "RDPTR_KEY" -> 165 
200 - Debug: "KEY1" -> 1164378403 
200 - Debug: "KEY2" -> 3455027627 
201 - Debug: "OPTKEY1" -> 135866939 
201 - Debug: "OPTKEY2" -> 1281191551 
201 - Debug: "SR_BSY" -> 0 
201 - Debug: "SR_PER" -> 2 
201 - Debug: "SR_EOP" -> 5 
201 - Debug: "CR_PG" -> 0 
201 - Debug: "CR_PER" -> 1 
201 - Debug: "CR_MER" -> 2 
201 - Debug: "CR_STRT" -> 6 
201 - Debug: "CR_LOCK" -> 7 
201 - Debug: "CR_PGSIZE" -> 8 
201 - Debug: "ACR_OFFSET" -> 0 
201 - Debug: "KEYR_OFFSET" -> 4 
201 - Debug: "OPT_KEYR_OFFSET" -> 8 
201 - Debug: "SR_OFFSET" -> 12 
201 - Debug: "CR_OFFSET" -> 16 
201 - Debug: "AR_OFFSET" -> 20 
201 - Debug: "OBR_OFFSET" -> 28 
201 - Debug: "WRPR_OFFSET" -> 32 
201 - Debug: "devices_default" -> "flash_base" 
201 - Debug: "devices_default" -> "sram_base" 
201 - Debug: "devices_default" -> "buffer_size" 
201 - Debug: "devices_default" -> "flash_size" 
201 - Debug: "devices_default" -> "flash_pgsize" 
201 - Debug: "device" -> "STM32L03xx" -> "core_id" 
202 - Debug: "device" -> "STM32L03xx" -> "chip_id" 
202 - Debug: "device" -> "STM32L03xx" -> "flash_size_reg" 
202 - Debug: "device" -> "STM32L03xx" -> "flash_int_reg" 
202 - Debug: "device" -> "STM32L03xx" -> "buffer_size" 
202 - Debug: "device" -> "STM32L03xx" -> "loader" 
202 - Debug: "device" -> "STM32L05xx" -> "core_id" 
202 - Debug: "device" -> "STM32L05xx" -> "chip_id" 
202 - Debug: "device" -> "STM32L05xx" -> "flash_size_reg" 
202 - Debug: "device" -> "STM32L05xx" -> "flash_int_reg" 
202 - Debug: "device" -> "STM32L05xx" -> "buffer_size" 
202 - Debug: "device" -> "STM32L05xx" -> "loader" 
202 - Debug: "device" -> "STM32L07xx" -> "core_id" 
202 - Debug: "device" -> "STM32L07xx" -> "chip_id" 
203 - Debug: "device" -> "STM32L07xx" -> "flash_size_reg" 
203 - Debug: "device" -> "STM32L07xx" -> "flash_int_reg" 
203 - Debug: "device" -> "STM32L07xx" -> "buffer_size" 
203 - Debug: "device" -> "STM32L07xx" -> "loader" 
203 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "core_id" 
203 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "chip_id" 
203 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "flash_size_reg" 
203 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "flash_int_reg" 
203 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "buffer_size" 
203 - Debug: "device" -> "STM32L1xx (Low/Med Density)" -> "loader" 
203 - Debug: "device" -> "STM32L1xx (High Density)" -> "core_id" 
203 - Debug: "device" -> "STM32L1xx (High Density)" -> "chip_id" 
203 - Debug: "device" -> "STM32L1xx (High Density)" -> "flash_size_reg" 
203 - Debug: "device" -> "STM32L1xx (High Density)" -> "flash_int_reg" 
203 - Debug: "device" -> "STM32L1xx (High Density)" -> "buffer_size" 
203 - Debug: "device" -> "STM32L1xx (High Density)" -> "loader" 
[... 211 lines cut ...]
6008 - Debug: ***[ void stlinkv2::setExitModeDFU() ]*** 
6012 - Info: "Changing mode to SWD..." 
6012 - Debug: ***[ void stlinkv2::setModeSWD() ]*** 
6012 - Debug: ***[ quint8 stlinkv2::getMode() ]*** 
6025 - Debug: ***[ void stlinkv2::setExitModeDFU() ]*** 
6134 - Info: "Fetching mode..." 
6134 - Debug: ***[ quint8 stlinkv2::getMode() ]*** 
6145 - Info: "Mode: Debug" 
6146 - Info: "Fetching status..." 
6146 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
6154 - Info: "Status: Core Halted" 
6155 - Info: "Fetching MCU Info..." 
6155 - Debug: ***[ quint32 stlinkv2::getCoreID() ]*** 
6161 - Info: CoreID: BC11477
6162 - Debug: ***[ void stlinkv2::resetMCU() ]*** 
6239 - Debug: ***[ quint32 stlinkv2::getChipID() ]*** 
6239 - Debug: ***[ qint32 stlinkv2::readMem32(QByteArray*, quint32, quint16) ]*** "Reading at 40015800" 
6246 - Info: CM0+ Searching at 40015800
6246 - Info: ChipID: 0x417
6246 - Debug: Looking for: 0x417
6246 - Debug: Found chipID
6246 - Info: Device type:  "STM32L05xx" 
6247 - Debug: ***[ quint32 stlinkv2::readFlashSize() ]*** 
6247 - Debug: ***[ qint32 stlinkv2::readMem32(QByteArray*, quint32, quint16) ]*** "Reading at 1FF8007C" 
6254 - Info: Flash size: 64 KB

fpoussin, 2015-12-11:
Perfect, I am closing the ticket.

Please re-open it if you encounter any other issues.

closed 2015-12-11 by fpoussin