fpoussin/QStlink2 #19: Unable to handle filenames without path information on linux

Lanchon, 2015-11-27:
platform: linux mint 17

```
$ qstlink2 -cr xxx
Invalid path: "xxx" 
"

This application can be used through the GUI or via the command line. (CLI) 

Command line usage: 
qstlink2 [-hqvcwreV] [path]

-h, --help: shows this help
-q, --quiet: Quiet output (Nothing) 
-v, --verbose: Verbose output (Debug) 
-c,--cli: Enable command line mode 
-w,--write: Enable write mode - Needs Path 
-r,--read: Enable read mode - Needs Path 
-e,--erase: Flash will be completely erased 
-V, --verify: When used with write, the file will be verified against flash 
-R, --reset: will reset the MCU 

Example to write and verify the device: 
qstlink2 -cwV file.bin
or 
qstlink2 --cli --write --verify file.bin 

More info here

" 
Version: 1.2.0 
```
```
$ qstlink2 -cr ./xxx
Verbose level: 3 
86 - Info: Devices list loaded. 
88 - Info: "29 Device descriptions loaded." 
88 - Info: File Path: "./xxx" 
88 - Info: Erase: false 
88 - Info: Write: false 
88 - Info: Verify: false 
88 - Info: "Searching Device..." 
388 - Info: libusb_bulk_transfer Error reading:  -7 
388 - Info: LIBUSB_ERROR_TIMEOUT 
389 - Info: "ST Link V2 / Nucleo found!" 
389 - Info: "Fetching version..." 
390 - Info: "Changing mode to SWD..." 
491 - Info: "Fetching mode..." 
492 - Info: "Mode: Debug" 
493 - Info: "Fetching status..." 
494 - Info: "Status: Core Running" 
494 - Info: "Fetching MCU Info..." 
495 - Info: CoreID: "2ba01477" 
497 - Info: CM3/4 Searching at "e0042000" 
497 - Info: ChipID: "422" 
497 - Info: Device type:  "STM32F30x" 
498 - Info: Flash size: 256 KB 
499 - Info: "Saving to ./xxx" 
511 - Info: Reading from "8000000" to "10000000" 
811 - Info: libusb_bulk_transfer Error reading:  -7 
811 - Info: LIBUSB_ERROR_TIMEOUT 
873 - Info: Progress: "1%" 
891 - Info: Progress: "2%" 
909 - Info: Progress: "3%" 
945 - Info: Progress: "4%" 
963 - Info: Progress: "5%" 
[...]
```

fpoussin, 2015-11-30:
Which Qt version are you using?

I need to run some tests but as far as I know it should work fine.

Lanchon, 2015-11-30:
This was version 1.2.0 installed from your PPA. Now I have master built against Qt 5, so I can't test this again.

I don't know which version of Qt I was using because I don't know if your PPA binaries are built against Qt 4 or 5.

If Qt 4 -> version is 4:4.8.5+git192-g085f851+dfsg-2ubuntu4.1
If Qt 5 -> version is 5.2.1+dfsg-1ubuntu14.3

Thanks!

fpoussin, 2015-11-30:
Okay, I will fix this, it doesn't really make sense to have a mandatory path.

Lanchon, 2015-11-30:
Lol, agree :)

fpoussin, 2015-11-30:
Fixed :)

closed 2015-11-30 by fpoussin