fpoussin/QStlink2 #20: Unable to program and erase on linux

Lanchon, 2015-11-27:
platform: linux mint 17
board: stm32f3discovery

`$ qstlink2 -cwVv ./stock-demo.bin >write.log 2>&1`
result: the first 0x800 bytes of flash are erased to 0xFF.

`$ qstlink2 -cev >erase.log 2>&1`
result: flash not erased.

see attached logs.
[write.log.txt](https://github.com/fpoussin/QStlink2/files/45468/write.log.txt)
[erase.log.txt](https://github.com/fpoussin/QStlink2/files/45469/erase.log.txt)

Lanchon, 2015-11-27:
program and erase bugs verified on current master compiled against Qt 5.

fpoussin, 2015-11-30:
I will run some tests. Are you sure your flash option registers are clear ? (flash protection)

Lanchon, 2015-11-30:
Yes, I can program the device fine with st-flash but fails with your code. I tested this repeatedly. I'm using the stlink/v2 embedded in the F3 discovery board, and i have the latest stlink firmware loaded.

fpoussin, 2015-11-30:
Ok, I will test on my F3 Discovery board.
Thanks.

Lanchon, 2015-11-30:
This is the file I'm programming, flashes and runs fine with:
`st-flash --reset write stock-demo.bin 0x8000000`

It's the stock software preloaded on my F3DISCO (property of ST, license their site).

[stock-demo.zip.txt](https://github.com/fpoussin/QStlink2/files/47526/stock-demo.zip.txt) (remove the '.txt' extension)

fpoussin, 2015-11-30:
Can you try with an older version of QStlink2?

sudo apt-get install qstlink2=1.1.7

Lanchon, 2015-11-30:
Sorry I'm unable to test at this time. Were you not able to reproduce?

fpoussin, 2015-11-30:
I can reproduce with 1.2.0, I'm working on a fix.

fpoussin, 2015-11-30:
For some reason if fails with your file, but other projects compiled with GCC are working fine.

Lanchon, 2015-11-30:
FYI, this file contains the stock firmware as shipped by ST and downloaded by your own tool (using PPA version 1.2.0). It can be successfully reprogrammed into the device (and then runs fine) with st-flash.

Try flashing some GCC project (works according to you), then download, then reflash. The same might probably happen. The download contains the whole flash (even unused parts). The bug might be related to the bin file being large or equal the size of the flash.

fpoussin, 2015-11-30:
I found the bug, it was trying to delete an extra flash page at the end of the flash area, so it was deleting the first one instead... fixed :)

closed 2015-11-30 by fpoussin

Lanchon, 2015-12-01:
Thank you! What about erase not erasing, was it the same bug and also fixed?

fpoussin, 2015-12-01:
I will check this one and reopen if needed. Thanks.

fpoussin, 2015-12-01:
I fixed the mass erase as well.

Lanchon, 2015-12-01:
Cool!