fpoussin/QStlink2 #24: Compile issue with libusb-1.0-0-dev on Ubuntu 14.04LTS

mrmowgli, 2015-12-03:
Using  QT4: 

    ~/dev/QStlink2$ qmake-qt4 --version
    QMake version 2.01a
    Using Qt version 4.8.6 in /usr/lib/x86_64-linux-gnu

And gcc:

    ~/dev/QStlink2$ gcc  --version
    gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4

Compile error: 

     ~/dev/QStlink2$ make
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o main.o src/main.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o mainwindow.o src/mainwindow.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o stlinkv2.o src/stlinkv2.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o devices.o src/devices.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o dialog.o src/dialog.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o transferthread.o src/transferthread.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o loader.o src/loader.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o qbaseusb.o QtUsb/src/qbaseusb.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o qusbmanager.o QtUsb/src/qusbmanager.cpp
    g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -D__QSTL_VER__="\"1.2.1\"" -DQUSB_LIBRARY -DQLIBUSB -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -Iinc -IQtUsb/src -I. -I. -o qlibusb.o QtUsb/src/qlibusb.cpp
    QtUsb/src/qlibusb.cpp: In member function ‘virtual void QUsbDevice::flush()’:
    QtUsb/src/qlibusb.cpp:215:74: error: invalid use of member function (did you forget the ‘()’ ?)
         libusb_bulk_transfer(mDevHandle, mConfig.readEp, (uchar*)(buf.data()+read), 4096, &read_bytes, 25);
                                                                              ^
    QtUsb/src/qlibusb.cpp: At global scope:
    QtUsb/src/qlibusb.cpp:218:8: error: prototype for ‘qint64 QUsbDevice::read(QByteArray*, quint32)’ does not match any in class ‘QUsbDevice’
     qint64 QUsbDevice::read(QByteArray* buf, quint32 maxSize)
            ^
    In file included from QtUsb/src/qlibusb.cpp:1:0:
    QtUsb/src/qlibusb.h:27:12: error: candidate is: virtual qint32 QUsbDevice::read(QByteArray*, quint32)
         qint32 read(QByteArray* buf, quint32 maxSize);
                ^
    QtUsb/src/qlibusb.cpp:272:8: error: prototype for ‘qint64 QUsbDevice::write(const QByteArray*, quint32)’ does not match any in class ‘QUsbDevice’
     qint64 QUsbDevice::write(const QByteArray* buf, quint32 maxSize)
            ^
    In file included from QtUsb/src/qlibusb.cpp:1:0:
    QtUsb/src/qlibusb.h:28:12: error: candidate is: virtual qint32 QUsbDevice::write(const QByteArray*, quint32)
         qint32 write(const QByteArray* buf, quint32 maxSize);
                ^
    make: *** [qlibusb.o] Error 1

mrmowgli, 2015-12-04:
Looks like a duplicate of #22

fpoussin, 2015-12-07:
Closing duplicate

closed 2015-12-07 by fpoussin