fpoussin/QStlink2 #25: USB error: -2

pVit, 2016-01-18:
Hello,
I have an issue with app.

App window:
> 32 Device descriptions loaded.
> Nucleo inserted
> Searching Device...
> ST Link V2 / Nucleo not found or unable to access it.
> Did you install the udev rules ?
> USB error: -2

Debug:
> 442 - Info: Could not open the devices.xml file. Using internal data.
> 451 - Info: Devices list loaded.
> 460 - Info: "32 Device descriptions loaded." 
> 31223 - Info: "Nucleo inserted" 
> 39934 - Info: "Searching Device..." 
> 39934 - Info: Cannot open device 
> 39934 - Info: Cannot open device 
> 39934 - Info: "ST Link V2 / Nucleo not found or unable to access it." 
> 39934 - Info: "Did you install the udev rules ?" 
> 39935 - Info: "USB error: -2" 
> 471888 - Info: "Transfer Aborted" 

Application always writes this. I tried it on Win 7 64bit, Win 7 32bit, Ubuntu 14.04, Raspbian (Linux sys on Raspberry PI).
I tried Nucleo board and Stlink V2 board, too, but both of them have the same issue.

Thanks for help.

fpoussin, 2016-01-19:
Which version are you using?
You should not be able to get the same logs on all platform, can you post the correct logs?

Thanks

pVit, 2016-01-22:
Hello, I use version 1.2.2.
Ubuntu and Raspbian have the same logs, I will try win when I'm home.

Problem is in QtUsb, it's not so multiplatform as It should be. Program couldn't find windows and linux, he always type "Platform not supported!" and program cannot be compiled. So I manualy write linux or win code into "else", compile is now ok, but it doesn't work at all. It only gets and writes what type of stlink I have connected but that's all.

fpoussin, 2016-01-22:
Are you using minGW on windows? (not supported)
Also only 32bit is supported on windows so far.

It should work on Linux, if you have the correct udev rules. 
Just install the package if you don't want to bother.

It has not been tested on ARM since a long time.
I have made a request to add ARM builds to my PPA.

fpoussin, 2016-01-22:
I updated the PPA, it's now also built for armhf.
Update the package and you should be good to go.

fpoussin, 2016-01-23:
As for the "Platform not supported" error, you most likely have an issue with your Qt platform setup.

Which version of Qt are you using?

pVit, 2016-01-29:
Thanks for response, I will try armhf at work and I will write you. I use Qt 4.8.6 at RPi and 5.1.1. on Win.

pVit, 2016-03-07:
Sorry for long waiting, I had some home troubles.
I tried this updated version today, there is no problem with "Platform not supported" but there is still "USB error: -2" so I cannot connect device for more actions.

fpoussin, 2016-03-07:
Ok, can you paste the debug output?

pVit, 2016-03-09:
Still the same:

> Starting /home/pi/Desktop/QStlink2-master/qstlink2...
> QGtkStyle was unable to detect the current GTK+ theme.
> Verbose level: 3
> Version: 1.2.3
> 815 - Info: Could not open the devices.xml file. Using internal data.
> 825 - Info: Devices list loaded.
> 845 - Info: "32 Device descriptions loaded." 
> 7614 - Info: "Nucleo inserted" 
> 11563 - Info: "Searching Device..." 
> 11568 - Info: Cannot open device 
> 11572 - Info: Cannot open device 
> 11573 - Info: "ST Link V2 / Nucleo not found or unable to access it." 
> 11574 - Info: "Did you install the udev rules ?" 
> 11575 - Info: "USB error: -2" 
> 16139 - Info: "Transfer Aborted" 

Still problem with udev rules. I did steps in "Setting up udev rules".

> For convenience, you may install udev rules file, 49-stlinkv*.rules, located
> in the root of repository. You will need to copy it to /etc/udev/rules.d,
> and then either reboot or execute
> $ udevadm control --reload-rules
> $ udevadm trigger

Everything gone well, but...

> Udev will now create a /dev/stlinkv2_XX or /dev/stlinkv1_XX file, with the appropriate permissions.
> This is currently all the device is for, (only one stlink of each version is supported at 
> any time presently)

... this file (/dev/stlinkv2_XX or /dev/stlinkv1_XX) was not created and I don't know why. I did it without "sudo" and with it too but still nothing.

fpoussin, 2016-03-10:
The last package version already installs the udev rules. It should be present and work directly.

Does it work if you launch the app with sudo?
Which nucleo board are you using? Can you get the USB PID/VID with dmesg or lsusb?

Mercotui, 2016-04-12:
Hello, I have the same problem, and it works when I sudo it, the same goes for other aplications (openocd),
How do I fix this? This is probably not the intended usage?
(made today, with qt4 on fedora23)

fpoussin, 2016-04-14:
Actually, you just need to reload udev.

closed 2016-04-27 by fpoussin