fpoussin/QStlink2 #26: MCU Info Device not found in database

tlamb96, 2016-05-23:
I'm using an STM32f4 Discovery board. When I try to flash the board, everything is fine until I reach the line:
`Fetching` MCU Info...
Device not found in database!
Disconnecting...
Disconnected.`
Any help would be great.

fpoussin, 2016-05-25:
Hi,

Can you provide some debug output?

Thanks.

tlamb96, 2016-05-25:
Sorry I didn't mean to close the issue. The output is:
Verbose level: 3
Version: 1.2.3
274 - Info: Could not open the devices.xml file. Using internal data.
276 - Info: Devices list loaded.
279 - Info: "32 Device descriptions loaded." 
6282 - Info: "Searching Device..." 
6309 - Info: "ST Link V2 / Nucleo found!" 
6309 - Info: "Fetching version..." 
6315 - Info: "Changing mode to SWD..." 
6425 - Info: "Fetching mode..." 
6428 - Info: "Mode: Debug" 
6428 - Info: "Fetching status..." 
6432 - Info: "Status: Core Running" 
6432 - Info: "Fetching MCU Info..." 
6435 - Info: CoreID: 000
6441 - Error: Did not find chipID!
6442 - Info: "Device not found in database!" 
6442 - Error: Device not found in database!
6442 - Info: "Disconnecting..." 
6446 - Info: "Disconnected." 
Any help would be great.

fpoussin, 2016-05-25:
Which OS are you running?
Are you sure the SWD jumpers are in place on the board?

tlamb96, 2016-05-26:
Ubuntu 14.04 and both of the jumpers are on the connect to the STLink pins (CN3 I believe)

tlamb96, 2016-06-07:
My STM32F4 board was shipped with Read Out Protection to level 1 which disables any writing to the memory. As an STM32 novice, this took quite a lot of time to figure out but is now fixed.

closed 2016-06-07 by tlamb96

fpoussin, 2016-06-07:
Thanks for the info, I'll have a look at making some checks in the code from now on.