fpoussin/QStlink2 #27: Failure to flash on STM32F103RET7

Timvrakas, 2016-07-21:
System Connects correctly, but the flashing progress bar never starts to move. -v shows: 
```
20459 - Info: Loader ":/bin/loader_f1.bin" 
20459 - Debug: ***[ qint32 stlinkv2::writeMem32(quint32, const QByteArray&) ]*** "Writing 128 bytes to 0x20000000" 
20460 - Debug: ***[ qint32 stlinkv2::writeMem32(quint32, const QByteArray&) ]*** "Writing 128 bytes to 0x20000080" 
20462 - Debug: ***[ qint32 stlinkv2::writeMem32(quint32, const QByteArray&) ]*** "Writing 128 bytes to 0x20000100" 
20463 - Debug: ***[ qint32 stlinkv2::writeMem32(quint32, const QByteArray&) ]*** "Writing 128 bytes to 0x20000180" 
20465 - Debug: ***[ qint32 stlinkv2::writeMem32(quint32, const QByteArray&) ]*** "Writing 98 bytes to 0x20000200" 
20465 - Info: Data is not 32 bit aligned! Padding with 2 Bytes
20466 - Debug: ***[ bool stlinkv2::writeRegister(quint32, quint8) ]*** 
20467 - Debug: ***[ quint32 stlinkv2::readRegister(quint8) ]*** 
20467 - Debug: Set register 15 to 20000000
20467 - Debug: ***[ void stlinkv2::runMCU() ]*** 
20468 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
20468 - Info: "Loader uploaded" 
20568 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
20668 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
20769 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
20870 - Debug: ***[ quint8 stlinkv2::getStatus() ]*** 
```

then there are just endless getStatus()

Thanks

fpoussin, 2016-07-21:
Are you using a discovery board? If so, which one?

Timvrakas, 2016-07-21:
No, it is a custom application. I was able to get it to work with the
official ST-Link Windows utility, and with st-flash on Linux. To program I
am using a SWD dongle, and I only have the 4 basic pins broken out: SWD,
CLK, Vcc, Gnd. Hence there is no debug data if that makes any difference.

fpoussin, 2016-07-22:
I was talking about QSTlink2's debug output, you only posted part of it.

Is the reset pin wired to the debugger?

Timvrakas, 2016-07-22:
Wow, OK, I went back to get the full debug script, and now it works fine. Very odd, it was failing under the same conditions earlier.
Sorry to waste your time, I will come back if I figure out anything more.

closed 2016-07-22 by Timvrakas

fpoussin, 2016-07-22:
No worries, let me know if you have any issues.