fpoussin/QStlink2 #30: QSTLink2 Incorrectly Identifies Nucleo-144 F207GZ Board and Fails to Program it

RatGit, 2016-10-05:
Unfortunately, QSTLink2 incorrectly reports my Nucleo-144 F207GZ board as a "STM32F405/415/407/417x" and when I program it with my generated "BIN" file it fails to execute.

Note: The same "BIN" file successfully downloads and executes when using the Windows STM32 ST-Link utility, (on a separate Windows machine)

Do you have any suggestions with regards to the QSTLink2 tool for Linux and the Nucleo-144 F207GZ board?

FYI:
I looked at the QSTLink2 "devices.xml" file and temporarily changed the "chip_id" value to "0x413" for the "STM32F2xx" device and commented out the "STM32F405/415/407/417x" device entry. After this the QSTLink2 utility successfully identified my board but still failed to program it successfully.

fpoussin, 2016-10-24:
Did you use a release or code from the master branch?

closed 2017-01-12 by fpoussin

RatGit, 2017-09-07:
I used the PPA