fpoussin/QStlink2 #33: Fail to upload binary via STLink2-1 Nucleo Debuger

FlYnN91, 2016-12-07:
Hi,

I have found that the qstlink2 (v1.2.3) fails to upload binaries if the STLink2-1 version is 2.28.16. I've noticed that there is a closed ticket already and I did use the ppa to install the 1.2.3 but I believe the issue is still there. qstlink2 works fine (with the same binary file) if STLink2-1 version is 2.23.9. 
Also the STM32-Utility uploads the binary file with no problem if the STLink2-1 version is 2.28.16.

I was able to mitigate the issue by downgrading STLink2-1 to 2.23.9.

For future sake, I've attached my binary (blinky) for STM32F429ZI-Nucleo.
[Dummy_Blinky.zip](https://github.com/fpoussin/QStlink2/files/636189/Dummy_Blinky.zip)

fpoussin, 2017-01-09:
I'll run some tests, they probably changed the protocol somehow.

jano2358, 2017-01-09:
I come to say the same, for version 2.28.18.
An I made used the same 'solution' but with the version 2.27.15, and I saved a copy of this firmware on my github: https://github.com/jano2358/STM32Workout

Nimayer, 2017-01-14:
Hi,
After flashing an __stm32f401-nucleo__ and __stm32f469-disco__ (resp STLinkv2 and STLink v2-1) with qstlink2 (1.30) the mbed virtual drive functionality appears to be broken, in fact when a file is dragged to the drive, it is not flashed to the board.

The only way I found to restore arm mbed functionality is to flash a binary of choice with the [stm32 utility for windows](http://www.st.com/en/embedded-software/stsw-link004.html), that while flashing reports a *read-protection* activated.

fpoussin, 2017-01-15:
1.3.0 is not released yet.
Retry with 1.2.3

fpoussin, 2017-01-15:
I moved 1.3.x code to the dev branch since people were assuming it's stable.

fpoussin, 2017-01-16:
Was able to test 2.28.18, seems like the main loader loop does no run.

Nimayer, 2017-01-17:
I can confirm that version 1.2.3 works with both stm32f401-nucleo(2.27.15) and stm32f469-disco(2.27.15). Thanks for moving the unstable code to a new branch, this way the AUR package in Arch Linux installs a working version.

fpoussin, 2017-01-18:
Still working on it, they might have deprecated some functions from the V1 API.

diond31, 2017-06-27:
@fpoussin I confirmed issue with ST-LINK/V2-1 version 2.28.17 on Nucleo64-F091RC.  This issues was also found on the Mastering STM32 Forum [(link)](http://www.carminenoviello.com/2016/02/26/restore-st-link-interface-bad-update-2-26-15-firmware/) and provides a link to a downgraded version 2.24.11 of the the ST-LINK/V2-1 firmware [(dropbox link)](https://www.dropbox.com/s/bm0cq4qo1mui1t0/stsw-link007.zip?dl=0).

Downgrading to ST-LINK/V2-1 2.24.11 on Nucleo64-F091RC fixed issue and allowed .bin uploads to Nucleo64-F091RC.  It would be nice for others to share links to more recent versions of ST-LINK/V2-1 firmware that are proven to work, until a fix for the most recent firmware is completed, as ST doesn't provide links to older version of ST-LINK/V2-1 firmware on their website.

diond31, 2017-06-27:
@jano2358 Thanks for providing [(link)](https://github.com/jano2358/STM32Workout) to downgraded ST-LINK/V2-1 firmware version 2.27.15.  I confirmed that ST-LINK/V2-1 version 2.27.15 works on Nucleo64-F091RC and uploads without issue.

fpoussin, 2017-06-28:
Thanks @diond31.

jorgemzz, 2017-12-24:
Thanks @diond31. I was unable to flash an STM32f411 on Ubuntu virtual machine using STLink2-1 version 2.29, although there was not such issue on macOS. Now everything works alright on virtual environment using STLink2-1  2.24.11.

wencelewski, 2018-02-02:
Hi,
I'm trying to upload a .bin file to my STM32F401 but i'm unable due the fact it stays at 0%. I'm using STLink2-1 2.24.11 and I alse tried to use the QStlink2 from the dev branch. It worked one time only, right after I've downgraded the STLink2-1 firmware to 2.24.11. But after that it returned to the same problem.
STlink2-1 Firmware Version 2.24.11
Board: STM32F401RE Nucleo

denisn73, 2018-02-03:
Hi, freind!
I can't help solve this problem, it is better to ask for help to the original source (https://github.com/fpoussin/QStlink2)
In my project, I implement the firmware download to Russian microcontrollers MDR32

> Hi,
> I'm trying to upload a .bin file to my STM32F401 but i'm unable due the fact it stays at 0%. I'm using STLink2-1 2.24.11 and I alse tried to use the QStlink2 from the dev branch. It worked one [...]