fpoussin/QStlink2 #34: Nucleo L476 does not program

creatron, 2017-01-11:
I tried the 'Latest commit Oct 12, 2016 version on a Nucleo L476 board. All operational but the device does not program. Programming with debug level 3 shows

33 Device descriptions loaded.
Searching Device...
ST Link V2 / Nucleo found!
Fetching version...
Changing mode to SWD...
Fetching mode...
Mode: Debug
Fetching status...
Status: Core Running
Fetching MCU Info...
Sending /home/yyyk/QStlink2/xxx.bin
Loader uploaded
Loader settings OK
Unlock failed
Loader uploaded
Loader settings OK.

The same program was used with F151 and F03x and works perfectly.
I also can read the device, and verify the device.
I use Opensuse Leap 42.2 (64Bit)

creatron, 2017-01-11:
More Info
STLink Info Ver 2.0 JTag 27 SQ=WIM 15

Windows says
STLink Util 3.9.0 Driver USB 4.4.0. STLink CLI 2.50
Firmware V2.J27.M15

fpoussin, 2017-01-11:
Please retry with the latest release

creatron, 2017-01-12:
I do use the latest from git. I also tried the lastest STM stlink 4.00 firmware upgrade, and the special downgrade jano 2358 firmware (java jar file).
Note:
STM StLinkV2 Ver 4.00 (STSW-LINK009	1.02) downloaded from http://www.st.com/en/embedded-software/stsw-link009.html

fpoussin, 2017-01-12:
The latest release and latest commit are two different things, check the releases section. The master branch is not considered as fully stable and tested on each commit.

creatron, 2017-01-13:
Ok I ran the following script on Opensuse Leap 42.2. I am not a GIT guru  so please check my script as I am do not know whether I cloned the release or commit.

#!/bin/bash
git clone https://github.com/fpoussin/QStlink2.git
cd QStlink2
git submodule update --init
#qt4-qmake # or  qt5-qmake
qmake
make -j 8
#sudo make install  # Optional

//  ---------------------------------------------------------
// test
git pull 

says you are Already up-to-date.

//-----------------------------------------------------------
Running the compiled version :  I read the stm L476 and then tried to program it with the file read

33 Device descriptions loaded.
Searching Device...
ST Link V2 / Nucleo found!
Fetching version...
Changing mode to SWD...
Fetching mode...
Mode: Debug
Fetching status...
Status: Core Running
Fetching MCU Info...
Saving to /home/gerrie/workspace/creatron/git_stlink/QStlink2/koos.bin
Size: 1024KB
Sending /home/gerrie/workspace/creatron/git_stlink/QStlink2/koos.bin
Loader uploaded
Loader settings OK
Unlock failed
Loader uploaded
Loader settings OK

Did I do something wrong?

Thanks for your prompt replies

fpoussin, 2017-01-13:
That's no good, you are using the latest commit of the master branch, which is not stable.

You can either use the tag of the latest release or download it. (or even use the ppa)
Using git: git checkout v1.2.3
You check check the tags with: git tags

creatron, 2017-01-13:
Ok I did that and tag replies as follows

gerrie@opensusewin7:~/workspace/creatron/qstlink/QStlink2> git tag
1.1.2
1.2.0
R132
v1.0.0
v1.0.1
v1.0.2
v1.0.3
v1.1.1
v1.1.3
v1.1.4
v1.1.5
v1.1.6
v1.1.7
v1.2.1
v1.2.2
v1.2.3
 But I still have the problem

The console report v 1.3.0. and the gui write report
a. tried to program, remove USB cable. (see error -1) and replugged and
retry.
Note the disconnected state

But the device is not programmed (zero percent),

Kind Regards

-- 
Gerrie de Jager

(Quite people have the loudest minds)

Phone +27(0)11 024 3832
Phone +27(0)11 026 5631
Fax   086 552 2890
Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

fpoussin, 2017-01-13:
git tag is just to see the tags.
you need to enter "git checkout v1.2.3" as I wrote above.

creatron, 2017-01-13:
Yes I did that and after checkout I checked it again with git tag.

-- 
Gerrie de Jager

(Quite people have the loudest minds)

Phone +27(0)11 024 3832
Phone +27(0)11 026 5631
Fax   086 552 2890
Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

fpoussin, 2017-01-13:
git tag only lists the tags, it does not make any check.
You are still on master or the application would display v1.2.3

You can check by typing "git branch"
You should see this: "* (HEAD detached at v1.2.3)"

closed 2017-01-21 by fpoussin