fpoussin/QStlink2 #35: Flash size not detected on STM32F411RET

prrole, 2017-04-17:
Using 4936cd629abe0ae8f8b2d16b1ff245008368c8bd (current as of 2017-04-17).

Run/Halt/Reset MCU work as expected - but flashing/reading is no go as the size is wrong...

````
26870 - Debug: { bool MainWindow::connect() }
26871 - Info: Searching Device...
26871 - Debug: { void stlinkv2::flush() }
26900 - Info: ST Link V2 / Nucleo found!
26901 - Info: Fetching version...
26901 - Debug: { stlinkv2::STVersion stlinkv2::getVersion() }
26911 - Debug: API version: 2
26912 - Info: Fetching mode...
26912 - Debug: { quint8 stlinkv2::getMode() }
26923 - Info: Mode: Debug
26924 - Debug: { void stlinkv2::setExitModeDFU() }
26924 - Debug: { quint8 stlinkv2::getMode() }
26936 - Info: Changing mode to SWD...
26936 - Debug: { void stlinkv2::setModeSWD() }
26936 - Debug: { void stlinkv2::setExitModeDFU() }
26936 - Debug: { quint8 stlinkv2::getMode() }
27056 - Info: Fetching mode...
27056 - Debug: { quint8 stlinkv2::getMode() }
27062 - Info: Mode: Debug
27063 - Info: Fetching status...
27063 - Debug: { quint8 stlinkv2::getStatus() }
27063 - Debug: { quint32 stlinkv2::readDbgRegister(quint32) }
27071 - Debug: Status Reg: 0x00030003
27071 - Debug: "110000000000000011"
27071 - Info: Status: Core Halted
27071 - Info: Fetching MCU Info...
27071 - Debug: { quint32 stlinkv2::getCoreID() }
27083 - Info: CoreID: 2BA01477
27083 - Debug: { void stlinkv2::resetMCU() }
27095 - Debug: { quint32 stlinkv2::getChipID() }
27095 - Debug: { quint32 stlinkv2::readDbgRegister(quint32) }
27106 - Info: CM3/4 Searching at E0042000
27106 - Info: ChipID: 0x431
27106 - Debug: Looking for: 0x431
27106 - Debug: Found chipID
27106 - Info: Device type:  "STM32F411xC/E"
27107 - Debug: { quint32 stlinkv2::readFlashSize() }
27107 - Debug: { qint32 stlinkv2::readMem32(QByteArray*, quint32, quint16) } "Reading 4 bytes from 1FFF7A22"
27123 - Info: Flash size: 64715 KB
````

The st-info program detects it ok:
````
./st-info --probe
Found 1 stlink programmers
 serial: 563f7506493f48532831113f
openocd: "\x56\x3f\x75\x06\x49\x3f\x48\x53\x28\x31\x11\x3f"
  flash: 524288 (pagesize: 16384)
   sram: 131072
 chipid: 0x0431
  descr: F4 device (low power) - stm32f411re
````

fpoussin, 2017-04-17:
Current master branch code is not stable, Did you try the releases?
I'm working on the next release to handle latest firmware.

prrole, 2017-04-17:
Looks the same with version 1.2.3 I got via apt-get.

My setup works well when I connect directly to SWDIO/SWCLK pins on the STM32F407 discovery board - so I think it's something with the 411 chip.

fpoussin, 2017-04-17:
Okay, it's probably the flash size register address that is wrong.
I'll have to check the datasheet and correct devices.xml. I'm accepting PRs if you don't mind :)

prrole, 2017-04-17:
I'm a total n00b with stm32, I found this program as it's the one used in the rather excellent book
"Mastering STM32".  It is my own PCB btw, so it could of course be something there - but seems stable, and I'm able to flash and run things using the tool at https://github.com/texane/stlink which also detects the size correctly.

The address for flash size seems to be ok, section 24.2:
http://www.st.com/content/ccc/resource/technical/document/reference_manual/9b/53/39/1c/f7/01/4a/79/DM00119316.pdf/files/DM00119316.pdf/jcr:content/translations/en.DM00119316.pdf

Possiblysomething about the organization or variable sector sizes (16/32/128kb)?  Not sure if it's normal in other parts, see section 3.3 in PDF above.

prrole, 2017-04-17:
Got a little further, don't think it warrants a pull request and all that.  Issue is the 411Ret is not recognized as an F4.  Added it:
stlinkv2.cpp:
````
quint32 stlinkv2::readFlashSize()
{
    PrintFuncName();
    QByteArray buf;

    this->readMem32(&buf, mDevice->value("flash_size_reg"));
    mDevice->insert("flash_size", qFromLittleEndian<quint32>((uchar*)buf.data()));
    if (mChipId == STM32::ChipID::F4 || mChipId == STM32::ChipID::F4_HD|| mChipId == STM32::ChipID::F4_DE) {
        mDevice->insert("flash_size", mDevice->value("flash_size") >> 16);
    }
    else {
        mDevice->insert("flash_size", mDevice->value("flash_size")&0xFFFF );
    }
    qInfo("Flash size: %d KB", mDevice->value("flash_size"));
    return mDevice->value("flash_size");
}

bool stlinkv2::setSTRT()
{
    PrintFuncName();
    quint32 mask = 0;

    if (mChipId == STM32::ChipID::F4 || mChipId == STM32::ChipID::F4_HD|| mChipId == STM32::ChipID::F4_DE) {
        mask |= (1 << STM32::Flash::F4_CR_STRT);
    }
    else {
        mask |= (1 << STM32::Flash::CR_STRT);
    }
    return (this->writeFlashCR(mask, true) & mask) == mask;
}
````
and stlinkv2.h
````
    const quint32 F4_DE = 0x431; /*STM32F411xC/E */
````
With this, the size is read correctly and I can READ the mcu no problem.  Writing however, does not work properly, as it seems to pad the beginning with 0xFF for 16 bytes:
````
Expecting: 00 00 02 20 6D 02 00 08 71 02 00 08 73 02 00 08 89 02 00 08 8B 02 00 08 A1 02 00
Got:FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 00 02 20 6D 02 00 08 71 02 00 08 73 02 00 08 89 02 00 08 8B 02 00 08 A1 02 00
````

Sorry I can't take this one all the way home, I just got to the chapter about installing the tool chain in the book, so VERY inexperienced when in comes to this MCU.

fpoussin, 2017-04-18:
Alright, I'll have a look.
Thanks for the info!

fpoussin, 2017-04-27:
Which ST-Link firmware version are you using? 
This is probably related to #33.

prrole, 2017-05-01:
The write stuff is probably related yes.  I did a checkout and patched version 1.2.3, and writing looks better - but still fails verify.

ST-link reported as Version 2, JTAG version 28, SWIM version 7

fpoussin, 2017-05-02:
Yes it it related, i'm working on this already.

fpoussin, 2017-05-05:
Closing this as the issue has been fixed, see the other ticket for the write issue.

closed 2017-05-05 by fpoussin