fpoussin/QStlink2 #37: Nucleo-64 STM32F446RE not recogised

vk4tux, 2017-05-16:
Odroid ARM Ubuntu 14.04 LTS
GUI log ;

33 Device descriptions loaded.
Searching Device...
ST Link V2 / Nucleo found!
Fetching version...
Fetching mode...
Mode: Mass Storage
Changing mode to SWD...
Fetching mode...
Mode: Debug
Fetching status...
Status: Core Running
Fetching MCU Info...
Device not found in database!
Disconnecting...
Disconnected.

cmd verbose ;

root@xu4odroid:~# qstlink2 -vv
Verbose level: 3
Version: 1.2.4
315 - Info: Devices list loaded.
322 - Info: 33 Device descriptions loaded.
29108 - Info: Searching Device...
29109 - Info: Cannot open device. Error code: 0
29136 - Info: ST Link V2 / Nucleo found!
29136 - Info: Fetching version...
29138 - Info: Fetching mode...
29139 - Info: Mode: Debug
29141 - Info: Changing mode to SWD...
29243 - Info: Fetching mode...
29244 - Info: Mode: Debug
29245 - Info: Fetching status...
29246 - Info: Status: Core Running
29248 - Info: Fetching MCU Info...
29248 - Info: CoreID: 2BA01477
29249 - Info: CM3/4 Searching at E0042000
29250 - Info: ChipID: 0x421
29250 - Error: Did not find chipID!
29251 - Info: Device not found in database!
29251 - Error: Device not found in database!
29252 - Info: Disconnecting...
29254 - Info: Disconnected.

vk4tux, 2017-05-16:
As a reference, on same system discovery board is recognised ;

33 Device descriptions loaded.
Searching Device...
ST Link V2 / Nucleo found!
Fetching version...
Fetching mode...
Mode: Mass Storage
Changing mode to SWD...
Fetching mode...
Mode: Debug
Fetching status...
Status: Core Running
Fetching MCU Info...

MCU Type :     STM32F405/415/407/417x

vk4tux, 2017-05-17:
Looks like no-one is interested. Back to stlink-gui then

fpoussin, 2017-05-17:
18 hours... Ever heard of "vacations"?
The mcu is simply missing from the device list.

vk4tux, 2017-05-17:
Ok, I would like to test when it is added. Thankyou

vk4tux, 2017-05-17:
Thankyou for the update,.  There seems too be more to it, git pull & checkout res/devices.xml, remake etc, however the issue remains ;   

 <device type="STM32F446" regtype="STM32F4" coretype="CM4F">
      <core_id>0x2ba01477</core_id>
      <chip_id>0x421</chip_id>
      <flash_size_reg>0x1FFF7A22</flash_size_reg>
      <flash_int_reg>0x40023c00</flash_int_reg>
      <loader>loader_f4.bin</loader>
      <SR_BSY>0x10</SR_BSY>
      <CR_STRT>0x10</CR_STRT>
      <CR_LOCK>0x1F</CR_LOCK>
      <CR_SER>0x1</CR_SER>
      <CR_PGSIZE>0x8</CR_PGSIZE>
      <buffer_size>0x8000</buffer_size>
    </device>

is seen in res/devices.xml

but seemingly not loading

;

root@xu4odroid:~/Downloads/QStlink2# ./qstlink2 -v           
Verbose level: 3
Version: 1.2.4
328 - Info: Devices list loaded.
335 - Info: 33 Device descriptions loaded.
6919 - Info: Searching Device...
6919 - Info: Cannot open device. Error code: 0
6946 - Info: ST Link V2 / Nucleo found!
6947 - Info: Fetching version...
6949 - Info: Fetching mode...
6951 - Info: Mode: Debug
6953 - Info: Changing mode to SWD...
7055 - Info: Fetching mode...
7057 - Info: Mode: Debug
7057 - Info: Fetching status...
7060 - Info: Status: Core Running
7061 - Info: Fetching MCU Info...
7062 - Info: CoreID: 2BA01477
7066 - Info: CM3/4 Searching at E0042000
7066 - Info: ChipID: 0x421
7066 - Error: Did not find chipID!
7067 - Info: Device not found in database!
7068 - Error: Device not found in database!
7069 - Info: Disconnecting...
7073 - Info: Disconnected.

vk4tux, 2017-05-17:
Thankyou very much, all working well ;

http://vk4tux.duckdns.org/F0DEI/Qstlinkgui9.jpg

closed 2017-05-17 by vk4tux

fpoussin, 2017-05-17:
You're welcome.

BeauterFeroceQC, 2017-12-01:
Same thing here, I have the same Nucleo and my logs are the same. The device is not found in database.

NimaANZ, 2020-03-10:
I have the same problem, on Ubuntu 18.04. 
Nucleo-F446RE 
QSTLink2 version 1.2.3:

33 Device descriptions loaded.
Searching Device...
ST Link V2 / Nucleo found!
Fetching version...
Changing mode to SWD...
Fetching mode...
Mode: Debug
Fetching status...
Status: Unknown
Fetching MCU Info...
Device not found in database!
Disconnecting...
Disconnected.
----------
 Verbose level: 3
Version: 1.2.3
134 - Info: libpng warning: iCCP: known incorrect sRGB profile
146 - Info: Devices list loaded.
152 - Info: "33 Device descriptions loaded."
4400 - Info: "Searching Device..."
4400 - Info: Cannot open device
4429 - Info: "ST Link V2 / Nucleo found!"
4429 - Info: "Fetching version..."
4430 - Info: "Changing mode to SWD..."
4531 - Info: "Fetching mode..."
4532 - Info: "Mode: Debug"
4533 - Info: "Fetching status..."
4534 - Info: "Status: Unknown"
4534 - Info: "Fetching MCU Info..."
4534 - Info: CoreID: 2BA01477
4535 - Info: CM3/4 Searching at E0042000
4535 - Info: ChipID: 0x005
4535 - Error: Did not find chipID!
4536 - Info: "Device not found in database!"
4536 - Error: Device not found in database!
4536 - Info: "Disconnecting..."
4537 - Info: "Disconnected."
--------------
would you please send me the additional lines to update the devices.xml file.