fpoussin/QStlink2 #38: Nucleo64 STM32F091RC "Device not found in database"

diond31, 2017-06-01:
I would like to develop on the STM32 M0 line, but unfortunately QSTLink2 responds with "Device not found in database!" when trying to connect.

tadmorgan, 2017-06-22:
Same here:

3714 - Info: ChipID: 0x442
3715 - Error: Did not find chipID!

tadmorgan, 2017-06-23:
Is it as simple as putting this in the devices.xml file?

    <device type="STM32F09x" coretype="CM0">
      <core_id>0x0bb11477</core_id>
      <chip_id>0x442</chip_id>
      <flash_size_reg>0x1FFFF7CC</flash_size_reg>
      <flash_int_reg>0x40022000</flash_int_reg>
      <buffer_size>0x1000</buffer_size>
      <loader>loader_f0.bin</loader>
    </device>

diond31, 2017-06-23:
Tad, it is that "simple"...Brilliant!

[image]

It used to say 33 devices and connected resulted in "Device not found in database! Disconnecting... Disconnected..." message that so many others have experienced.  Thanks again Tad

vk4tux, 2017-06-23:
> image

Are the missing parameters required? I guess not ;

     <device type="STM32F09x" coretype="CM0">
       <core_id>0x0bb11477</core_id>
       <chip_id>0x442</chip_id>
       <flash_size_reg>0x1FFFF7CC</flash_size_reg>
       <flash_int_reg>0x40022000</flash_int_reg>
       <loader>loader_f4.bin</loader>
       <SR_BSY>0x10</SR_BSY>
       <CR_STRT>0x10</CR_STRT>
       <CR_LOCK>0x1F</CR_LOCK>
       <CR_SER>0x1</CR_SER>
       <CR_PGSIZE>0x8</CR_PGSIZE>
       <buffer_size>0x1000</buffer_size>
       <loader>loader_f0.bin</loader>
    </device>

On 24/06/17 01:57, diond31 wrote:

fpoussin, 2017-06-23:
Glad you could fix it by yourself! 
Can you submit a pull request to merge the fix?

Thanks

vk4tux, 2017-06-23:
I just asked a simple question if the missing parameters are required as 
for the format shown for the other types and I gave a guessed example.

I had a similar situation with the STM32F446RE a couple of months back,. 
Tads example for this one  lacks three lines seen in the others, is that 
important ?

fpoussin, 2017-06-23:
Yes that should be correct.

diond31, 2017-06-23:
fpoussin...I'll pull the devices.xml and supply a merge request.

vk4tux...I suspect in some cases they are, but I don't have a complete understanding of the xml file parameters as they are related to Qstlink2, the ARM family of architectures, or the device specific data sheets.

This fix was a cut and paste of the "STM32F07x" XML and changing with tadmorgan's fix (which I verified by watch the terminal window after running "qstlink2 -v").  I know the Nucelo-STM32F07x only has 128k of flash, so I expected the flash size to be wrong as the Nucleo-STM32F09x has 256k, but the screen shot above shows that Qstlink2 found all 256k.

fpoussin, 2017-06-23:
Flash size is detect at runtime by reading a register, so it should always be correct.

You just need to make sure the buffer size is not bigger than the ram (0X1000 here, so 8Kb)

diond31, 2017-06-23:
reopen to pull request

closed 2017-06-23 by diond31

diond31, 2017-06-23:
@fpoussin the STM32F09x has 32Kb of SRAM.  Would change <buffer_size>0x8000</buffer_size> allow access to all 32Kb of SRAM?

fpoussin, 2017-06-24:
Yes