fpoussin/QStlink2 #43: STM23L0 discovery board, STLinkv2 not found

rajesh0therascal, 2018-03-15:
I am getting this error when i try to connect my STM32L0 discovery board

Log:
Searching Device...
ST Link V2 / Nucleo not found or unable to access it.
Did you install the udev rules ?
USB error: -2

Pls help.

fpoussin, 2018-03-26:
Hi,

You need the chip id to add it to the devices.xml file.

jiapei100, 2018-05-28:
Same problem here. I'm using [Nucleo-F767ZI ](http://www.st.com/en/evaluation-tools/nucleo-f767zi.html) . It seems [Nucleo-F767ZI ](http://www.st.com/en/evaluation-tools/nucleo-f767zi.html) is NOT supported by QStlink2 yet?

> 35 Device descriptions loaded.
> Searching Device...
> ST Link V2 / Nucleo found!
> Fetching version...
> Fetching mode...
> Mode: Debug
> Changing mode to SWD...
> Fetching mode...
> Mode: Debug
> Fetching status...
> Status: Core Running
> Fetching MCU Info...
> Device not found in database!
> Disconnecting...
> Disconnected.
> Searching Device...
> ST Link V2 / Nucleo found!
> Fetching version...
> Fetching mode...
> Mode: Debug
> Changing mode to SWD...
> Fetching mode...
> Mode: Debug
> Fetching status...
> Status: Core Running
> Fetching MCU Info...
> Device not found in database!
> Disconnecting...
> Disconnected.

Cheers
Pei