fpoussin/QStlink2 #45: Device not found in database!  - Nucleo-F767ZI, STM32F767ZI

jiapei100, 2018-05-28:
I'm using [Nucleo-F767ZI ](http://www.st.com/en/evaluation-tools/nucleo-f767zi.html) . It seems [Nucleo-F767ZI ](http://www.st.com/en/evaluation-tools/nucleo-f767zi.html) is NOT supported by QStlink2 yet?

> 35 Device descriptions loaded.
> Searching Device...
> ST Link V2 / Nucleo found!
> Fetching version...
> Fetching mode...
> Mode: Debug
> Changing mode to SWD...
> Fetching mode...
> Mode: Debug
> Fetching status...
> Status: Core Running
> Fetching MCU Info...
> Device not found in database!
> Disconnecting...
> Disconnected.

Cheers
Pei

fpoussin, 2018-08-03:
Hi,
You can add an entry in devices.xml with the MCU information to enable support.
Thanks

bm16ton, 2021-07-15:
this worked for me, tho it reads 2025k.not 2048k  It was reading random 128k to 320k, despite multiple cleans and compiles, for sum reason retyping in the 0x1FF0F442 then clean and compile its stable at 2025k, totally confused about that.

    <device type="STM32F767" regtype="STM32F4" coretype="CM7">
      <core_id>0x5BA02477</core_id>
      <chip_id>0x451</chip_id>
      <flash_size_reg>0x1FF0F442</flash_size_reg>
      <flash_int_reg>0xE0042004</flash_int_reg>
      <loader>loader_f4.bin</loader>
      <SR_BSY>0x10</SR_BSY>
      <CR_STRT>0x10</CR_STRT>
      <CR_LOCK>0x1F</CR_LOCK>
      <CR_SER>0x1</CR_SER>
      <CR_PGSIZE>0x8</CR_PGSIZE>
      <buffer_size>0x8000</buffer_size>
    </device>